travisgoodspeed/goodwatch #58: Board thickness is undocumented

jaseg, 2017-12-21:
Awesome project! What thickness are the PCBs though? Standard 1.6mm sounds a bit much to me. I think it would make sense to list this in the BOM.

travisgoodspeed, 2017-12-21:
It would.  I'll check my receipts and close this issue when it's documented.

err4nt, 2018-01-01:
Greetings neighbors, I opened up my 3208 and measured the board thickness at 0.5mm. The picture on https://github.com/travisgoodspeed/goodwatch/wiki/GoodWatch10 has them looking very close in thickness.

travisgoodspeed, 2018-01-07:
I manufactured in .024" (~0.61mm), but 0.5mm would fit better if your manufacturer allows it.

Close this issue when the documentation is updated.

etbusch, 2018-01-30:
My boards manufactured in 0.6mm fit snugly in the enclosure of a CA53W-1 watch. Since 0.6mm thickness is the thinnest commonly available from cheap PCB fabs, perhaps it should be the recommended value for this project.

travisgoodspeed, 2018-02-06:
Fixed in 37d6f73 with the GoodWatch21 schematic.

commit 37d6f73a (2018-02-06): Documents board thickness.  Close #58.
closed 2018-02-06 by travisgoodspeed