travisgoodspeed/goodwatch #69: Unable to set RTC

etbusch, 2018-01-31:
Hey Neighbors,

I'm having an issue with the set function of the clock app. I'm not sure if it's related to the chip errata regarding writing to the RTC, the applet software, or an issue with my particular board. However I've verified that the following happens consistently:

- Hitting the Program button doesn't consistently cause the watch to enter set mode, and sometimes freezes the updating of the clock app

- If I can enter set mode, I can cycle through the fields, but setting any of them causes the watch to lock up and eventually reboot.

If any of you could try to reproduce this, I'd appreciate it. I'll be assembling another board this evening to make sure that it isn't an issue with the soldering or assembly, and if it fixes it, I will close this issue.

Thanks!

ea, 2018-01-31:
I've had a similar issue (lock-up when trying to set the clock and occasiona boot stuck on key init) on at least two boards.

Could I suggest you go back a couple of revisions ? This started happening to me after key interrupts change ( in 6ba460e ). Does going back to 3afe3e464971ea126f91f5995d2257d55850745e maybe solve this ? 

Maybe the added key interrupt handling code makes some otherwise unnoticeable board or soldering deficency more prominent?

etbusch, 2018-01-31:
I too sometimes get stuck in a loop during the keypad init sequence. I'll hop back to the commit before the interrupt changes to see if it is reproduced.

travisgoodspeed, 2018-01-31:
Looks like I screwed something up with recent patches.  I don't think this is the RTC errata, unless you are using a different compiler.

Hunting the problem down now.

travisgoodspeed, 2018-01-31:
@etbusch Can you reliably enter the programming mode by pressing `+` and `-` at the same time?

travisgoodspeed, 2018-01-31:
It should be a lot better now, but I'll hold off on closing this issue until the debouncing is a little bit better.

etbusch, 2018-02-01:
Thanks for the prompt responses Travis!

The latest master absolutely fixes the bugs I described. The program button is a little touchy still, especially with the small inset button of the CA53W-1, but it's much better, and I can set the watch!

We could improve the bouncing in and out of time set mode just by not using the program button as a toggle, and making the user complete the set process to return to the clock.

````
  /* The SET button will move us into the programming mode. */
  if(sidebutton_set()){
    settingclock=!settingclock;
  }
````

to

```
 /* The SET button will move us into the programming mode. */
  if(sidebutton_set()){
    settingclock=1;
  }
```

travisgoodspeed, 2018-02-01:
I think I'll make the side buttons latching for convenience, rather than require a complete tour of the time and date.

Closing issue.  Will reopen if there are further problems.

closed 2018-02-01 by travisgoodspeed