travisgoodspeed/goodwatch #77: Buzzer Support

travisgoodspeed, 2018-02-07:
Use this issue to track buzzer support on P2.7.

travisgoodspeed, 2018-02-08:
Looks like driving the piezo directly from an IO pin wasn't such a great idea.  We won't have functional buzzer hardware until the GW21.

etbusch, 2018-02-08:
What's the issue? Not enough drive current?

travisgoodspeed, 2018-02-08:
Too much current, or too much reflected current, I think.  Whenever I set P2.7 both OUT and high, it glitches into a reboot.

The Casio board uses a transistor to drive the piezo, but I'll try a series resistor in the morning to keep the BOM small.  You can still get a bit of noise by flipping 2.7 between IN and OUT with the output low, but it doesn't sound very nice or loud.

travisgoodspeed, 2018-02-08:
If you'd care to repeat my crash, just uncomment the initialization of the `buzz` module in `main.c`.

etbusch, 2018-02-08:
I'll give this a shot over the weekend. I imagine the reason that the piezo on the original board has a drive transistor and the fairly large inductor in parallel is that the oscillation of the piezo has enough of an effect on the power rails that it can glitch the clock and the chip.

I don't have the equipment to measure the inductor on the original board, but I wonder if a SMD equivalent can be sourced.

travisgoodspeed, 2018-02-08:
So this is what the Chronos watch uses, where Q1 is a BC846 NPN transistor, R4 is 47k, and L1 is 32mH (Johanson L-07C12NJV4).  Sorry for the image quality; the Chronos schematics are only included as a blurry JPEG in the app notes.

[image]

Still hoping that a simple series resistor will suffice, but we'll see how it goes.

etbusch, 2018-02-10:
I just got my test watch's buzzer wired up (the case is not assembled, and the buzzer has short wires running to it), and it seems to be working just fine. Once I have the alarm functionality in a good place I'll program my other watch and see if the buzzer works inside the case, and update this issue accordingly.

travisgoodspeed, 2018-02-10:
Weird, but excellent news.  Perhaps I shorted a wire somewhere?

I'll try to confirm your results before finalizing the GW21 gerbers.

etbusch, 2018-02-10:
I've confirmed the buzzer working on a second watch. It can do most tones, but some end up sounding a bit funky, maybe due to timing.

I'll be sending a pull request of my alarm functionality shortly, with some extra buzzer code included.

travisgoodspeed, 2018-02-17:
I've confirmed a second watch which bootloops around the buzzer initialization when the buzzer spring is attached.  What could we be doing differently, to see such wildly different results?  Have any of the rest of you fine folks replicated either @etbusch's success or my failure?

etbusch, 2018-02-17:
I will be assembling a third watch for a friend tomorrow and will see what that one does.

etbusch, 2018-02-18:
The third watch I've put together has a working buzzer. All three of these are the CA-53W model, maybe the other cases has different buzzer elements?

travisgoodspeed, 2018-02-18:
I've only been using the 506 case lately.  Will try a 53 early next week.

closed 2018-04-22 by travisgoodspeed