travisgoodspeed/goodwatch #83: Building/linking firmware fails

teon, 2018-03-29:
I have an error during linking stage when building firmware:

```
10:21:14 teon@x firmware ±|master|→ make
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o rtcasm.o rtcasm.S
echo "#define GITTAG" 0x`git describe --always` >gittag.h
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o main.o main.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o lcd.o lcd.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o lcdtext.o lcdtext.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o rtc.o rtc.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o keypad.o keypad.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o apps.o apps.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o applist.o applist.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o sidebutton.o sidebutton.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o power.o power.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o uart.o uart.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o monitor.o monitor.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o ucs.o ucs.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o buzz.o buzz.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o radio.o radio.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o packet.o packet.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o dmesg.o dmesg.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o codeplug.o codeplug.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o rng.o rng.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o libs/assembler.o libs/assembler.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o libs/morse.o libs/morse.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o apps/clock.o apps/clock.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o apps/alarm.o apps/alarm.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o apps/rpn.o apps/rpn.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o apps/hex.o apps/hex.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o apps/stopwatch.o apps/stopwatch.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o apps/phrase.o apps/phrase.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o apps/rngapp.o apps/rngapp.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o apps/tuner.o apps/tuner.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o apps/morse.o apps/morse.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o apps/beacon.o apps/beacon.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o apps/iclicker.o apps/iclicker.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os    -c -o apps/submenu.o apps/submenu.c
msp430-gcc -mmcu=cc430f6137 -Wall -I. -Os -T msp430.x -o goodwatch rtcasm.o main.o lcd.o lcdtext.o rtc.o  keypad.o apps.o applist.o sidebutton.o power.o uart.o monitor.o ucs.o buzz.o radio.o packet.o dmesg.o codeplug.o rng.o libs/assembler.o libs/morse.o  apps/clock.o apps/alarm.o apps/rpn.o apps/hex.o apps/stopwatch.o apps/phrase.o apps/rngapp.o apps/tuner.o apps/morse.o apps/beacon.o apps/iclicker.o apps/submenu.o
/usr/bin/msp430-ld: goodwatch section `.rodata' will not fit in region `rom'
/usr/bin/msp430-ld: section .vectors loaded at [000000000000ff80,000000000000ffff] overlaps section .rodata loaded at [000000000000d294,000000000001015d]
/usr/bin/msp430-ld: region `rom' overflowed by 510 bytes
/usr/lib/gcc/msp430/4.6.3/mcpu-430x/mmpy-16/libcrt0.a(_copy_data.o): In function `__do_copy_data':
(.init4+0x12): relocation truncated to fit: R_MSP430_16_BYTE against symbol `__data_load_start' defined in *ABS* section in goodwatch
apps/iclicker.o: In function `tune':
iclicker.c:(.text+0x12): relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
apps/submenu.o: In function `submenu_draw':
submenu.c:(.text+0xe): relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
apps/submenu.o: In function `submenu_drawselected':
submenu.c:(.text+0x8e): relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
/usr/lib/gcc/msp430/4.6.3/mcpu-430x/mmpy-16/libgcc.a(_addsub_sf.o): In function `_fpadd_parts':
/tmp/packerbuild-1000/gcc-msp430/gcc-msp430/src/gcc-4.6.3/build/msp430/mcpu-430x/mmpy-16/libgcc/../../../.././gcc/fp-bit.c:623: relocation truncated to fit: R_MSP430_16_BYTE against symbol `__thenan_sf' defined in .rodata section in /usr/lib/gcc/msp430/4.6.3/mcpu-430x/mmpy-16/libgcc.a(_thenan_sf.o)
/usr/lib/gcc/msp430/4.6.3/mcpu-430x/mmpy-16/libgcc.a(_mul_sf.o): In function `_fpmul_parts':
/tmp/packerbuild-1000/gcc-msp430/gcc-msp430/src/gcc-4.6.3/build/msp430/mcpu-430x/mmpy-16/libgcc/../../../.././gcc/fp-bit.c:811: relocation truncated to fit: R_MSP430_16_BYTE against symbol `__thenan_sf' defined in .rodata section in /usr/lib/gcc/msp430/4.6.3/mcpu-430x/mmpy-16/libgcc.a(_thenan_sf.o)
/usr/lib/gcc/msp430/4.6.3/mcpu-430x/mmpy-16/libgcc.a(_thenan_sf.o):(.debug_info+0x13c): relocation truncated to fit: R_MSP430_16_BYTE against symbol `__thenan_sf' defined in .rodata section in /usr/lib/gcc/msp430/4.6.3/mcpu-430x/mmpy-16/libgcc.a(_thenan_sf.o)
/usr/lib/gcc/msp430/4.6.3/mcpu-430x/mmpy-16/libgcc.a(_clzsi2.o): In function `__clzsi2':
/tmp/packerbuild-1000/gcc-msp430/gcc-msp430/src/gcc-4.6.3/libgcc/../gcc/libgcc2.c:709: relocation truncated to fit: R_MSP430_16_BYTE against symbol `__clz_tab' defined in .rodata section in /usr/lib/gcc/msp430/4.6.3/mcpu-430x/mmpy-16/libgcc.a(_clz.o)
/usr/lib/gcc/msp430/4.6.3/mcpu-430x/mmpy-16/libgcc.a(_clz.o):(.debug_info+0x98): relocation truncated to fit: R_MSP430_16_BYTE against symbol `__clz_tab' defined in .rodata section in /usr/lib/gcc/msp430/4.6.3/mcpu-430x/mmpy-16/libgcc.a(_clz.o)
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/lib/mcpu-430x/mmpy-16/libc.a(vuprintf.o): In function `vuprintf':
/tmp/packerbuild-1000/msp430-libc/msp430-libc/src/msp430-libc-20120224/src/./stdlib/vuprintf.c:433: relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
collect2: ld returned 1 exit status
make: *** [Makefile:26: goodwatch] Error 1
```

I'm using ArchLinux with the following GCC for MSP430:
https://sourceforge.net/projects/mspgcc/

(aur packages: gcc-msp430 msp430mcu msp430-jtag)

Any idea if it's my compilation setup or a bug in source?

ea, 2018-03-29:
Yep, I ran into that myself this morning. With added alarm code and apps it was indeed pushed over ROM capacity. Phrase app is obviously the biggest hog. 

Simple temporary solution is to remove the apps you don't want from the makefile and from applist.c.  I'll try to make a makefile configurable easily ...

travisgoodspeed, 2018-03-29:
Closed with the recent merge.

closed 2018-03-29 by travisgoodspeed