travisgoodspeed/goodwatch #85: OOK Emulation Example

travisgoodspeed, 2018-04-04:
Use this issue to track development of a quick emulator for these cheap OOK remote controls, including support for the full power table and safety checks to identify packet frame under/overflows.

[image]

travisgoodspeed, 2018-04-05:
The OOK example is now fully functional.  I'll close this issue after documented it in the Wiki.

travisgoodspeed, 2018-04-06:
Documentation is at https://github.com/travisgoodspeed/goodwatch/wiki/OOK_Example

I'll add a few screenshots, then close this issue.

etbusch, 2018-04-06:
Awesome! I'm excited to dig into this, and to develop new applications once my LimeSDR mini comes in to replace a failed RTL-SDR. Thanks for keeping the momentum moving forward on this project!

closed 2018-04-07 by travisgoodspeed