travisgoodspeed/goodwatch #95: GoodWatch22 Hardware

travisgoodspeed, 2018-07-23:
The GoodWatch22 has no BOM changes from the '21, but provides for better ground-plane via stitching and other minor changes.  It also moves the antenna connector to the back, to ease routing of the antenna wire in assembly.

Use this issue to track that the design is:
- [ ] Drafted
- [ ] Gerbers have been finalized.
- [ ] Boards have been ordered.

After the GW22, we will widen the filters for the GW30.

etbusch, 2018-07-23:
Does the antenna connector on the back allow for the fitment of the Yageo antenna inside the watch case?

travisgoodspeed, 2018-07-23:
@etbusch Sadly not.  The Casio case doesn't leave enough internal room for the Yageo antenna, unless you provide some external source of power.

etbusch, 2018-07-24:
Gotcha. Have we considered the use of a helical pcb antenna (like the Yageo is) embedded into the board itself? I may take some measurements of the Yageo and see about adapting a similar design for a 0.6mm pcb and a more broadband application.

travisgoodspeed, 2018-07-24:
The 433MHz Yageo is far too thick to fit by cuts in the PCB, but a 900MHz antenna might.  Replacing the battery holder and back of the watch module might also provide sufficient room, but it will still be a tight fit.

If you'd care to help figure this out, let me know to spin up an issue for it.

travisgoodspeed, 2018-07-31:
Gerbers have been finalized and boards are on the way.  Closing this issue.

I'll spin up a new issue for the GW30, which will include a wider radio filter design.

closed 2018-07-31 by travisgoodspeed