travisgoodspeed/goodwatch #100: GoodWatch30 Hardware

travisgoodspeed, 2018-08-01:
Use this issue to track the design of the GW30, which will include wider filters and substantially different layout in Kicad 5.

travisgoodspeed, 2018-08-02:
This BOM has been updated, but backordered parts will take twelve weeks for delivery, so I won't be manufacturing PCBs for the next month or so.  That'll give me time to extra-super-double-check footprints and maybe drop the chip antenna for SMA or u.FL.

esden, 2018-08-18:
@travisgoodspeed are you still interested in the u.FL footprint instead of the chip antenna? We have enough space there to even add that footprint under the chip antenna footprint. This way we get all the options.

esden, 2018-09-02:
Ok created a pull request #104 just with the u.fl antenna footprint added. Let me know if you are OK with the placement. I would rather resolve that before I dive into more work that builds upon this commit.

I have more changes I would like to propose:
* we can use smaller via on the board to gain bit more routing space and cleaner copper pattern 0.6/0.4 are larger than the default I usually use. We could go to the still very common 0.5/0.3 via size.
* use 0.2 antipad and 0.2 spoke on the back side ground plane the same way as we have it on the front fill. (did not want to change that in case there is some technical reason for that. The only pad that probably should be connected with wide spokes is the battery terminal.
* round the board corners with 2.5mm radius to match the original board found inside the watch. This will make it easier to put it inside the original plastic molding.
* add 1BitSquared internal part tracking codes as custom flields (for example: `key: cap-cer-0402-1u`) to the schematic symbols to make it easier for me to order parts using my existing BOM ordering and P&P workflow.

Not sure what else but I will communicate it if I run into something else. :) I can have all these changes done by Tuesday depending on when I get the go ahead ;-)

travisgoodspeed, 2018-09-29:
I'm now wearing a GoodWatch30!  Happily closing this issue.

closed 2018-09-29 by travisgoodspeed