travisgoodspeed/goodwatch #121: GoodWatch31 Hardware

travisgoodspeed, 2019-09-01:
The GoodWatch31 will be in many ways identical to the '30, except for an experimental backlight.  There will be no significant changes from the BOM, and the LEDs themselves will be optional.

travisgoodspeed, 2019-09-01:
Rohm Semiconductor's [SML-P12VTT86R](https://www.digikey.com/product-detail/en/rohm-semiconductor/SML-P12VTT86R/846-1168-1-ND/5011427) device provides an 0402 LED with minimal height.

[image]

1divf, 2019-09-02:
Hi Travis,
a backlight would be great!
Some time ago I started to work on a backlight and did some measurements.
Maybe a slim side view LED could be placed within the small gap next to the LCD. 
[image]
The gap is only 0.6mm wide and 0.6mm wide side view LED are not that common :/

In addition a thigh fit might cause some stress on the solder joins when pressing a button. So removing a little bit of plastic and a very accurate placement of the LED might help at this point.

I created a 3D model and some svg files for a kicad fabrication layer.
[CA-53W_frame_3D.zip](https://github.com/travisgoodspeed/goodwatch/files/3565075/CA-53W_frame_3D.zip)
[fab_layer.zip](https://github.com/travisgoodspeed/goodwatch/files/3565009/fab_layer.zip)

Maybe these files will help to find the best spot for a tiny LED.

Best,
Ulf

travisgoodspeed, 2019-09-03:
@1divf These CAD files are excellent!  If you wouldn't mind, please send them as a pull request so that they can become a part of the repo?

travisgoodspeed, 2019-10-12:
To better fit the casing, the GW11 will be manufactured in 0.5 mm (0.020") thickness.

freefuel, 2020-08-31:
is there a preferred board house for the boards?

travisgoodspeed, 2020-09-24:
I use [Gold Phoenix PCB](https://www.goldphoenixpcb.com/).  Their website is a bit janky, but they've done good work since the days when you'd email gerbers and send payment by carrier pigeon.

vgnshlvnz, 2025-03-24:
Dear Travis, @travisgoodspeed 

I am new to this game and I would like hands on the board. If you have spare GoodWatch30/31 can you send me 1? I will pay for them. No issue.

freefuel, 2025-03-24:
Please tell me how you managed to send this message from Github

Thank you

Justin

travisgoodspeed, 2025-03-24:
I have plenty of boards to share.  Just email your shipping address my way.

--Travis