travisgoodspeed/goodwatch #123: Question about packet_tx()

notpike, 2019-10-01:
Hi!

I'm working on an TouchTunes wireless remote application and I have a question about the packet_tx() function. Looking at the sample code in ook.c packet_tx() is having an uint8_t casted char array piped in but I'm having some difficulty recreating it with a NEC encoder I'm building. The encoder I'm building creates a 16 byte message and returns a char array (returnValue) which is saved to another char array (button_array). I'm not sure if the whole 16byte value is being saved in the array or not and honestly my next move is to create a 3d uin8_t array but I would like to know what I'm doing wrong. :rofl:  

My experience with C is limited so any advice would be appreciated. Bellow is what I have done so far and thanks! Hopefully we can control jukeboxes with this watch soon! :smile:  

NEC Encoder: encode() : 108
Command Creation: build_button_array() : 208

https://gist.github.com/notpike/c2b0f53e68cd128e804020a4c080b9c0

travisgoodspeed, 2019-10-02:
Howdy @notpike!

It looks like `char *button_array[10];` is supposed to be an array of ten pointers to byte arrays, and that is well enough, except that you are populating the array with pointers to *stack* memory in `build_button_array()`.

```C
// Build Array
void build_button_array(int pin) {
	char returnValue[17];

	encode(returnValue, jukebox_commands[4],pin);  // Skip
	button_array[0] = returnValue;
```

Because this is stack memory, `char returnValue[17]` will have its contents overwritten by other functions, corrupting your packet before it is transmitted.  And when you uncomment the rest of that function, you are overwriting the same bytes with each additional call to `encode()`.

Instead of maintaining an array in RAM for every button, I'd recommend writing a function that returns the encoded packet in a `static` array, regenerating it only if the command or the pin changes.  This should use just twenty-two bytes of RAM, while still returning in an instant if the command and pin are the same as the last time it was called.

```C
// Build Array
uint8_t* build_jukebox_packet(int cmd, int pin) {
	static int lastpin=-1, lastcmd=-1;
	static uint8_t packet[17];  //Must be static so it isn't overwritten.
	
	//Update the packet only if the pin has changed.
	if(lastpin!=pin or lastcmd!=cmd)
		encode(packet, cmd, pin);  // Skip

	//Return our static packet value.
	return packet;
}
```

From Knoxville,
--Travis

notpike, 2019-10-03:
Well good news came with mix blessings lol. 

The encoder now works grate, how ever even number keys (0,2,4,6,8) begins to transmit however the message gets malformed. Viewing it with a spectrum analyzer it looks like the transmission gets cut short right at the beginning. Below is a fork of the current progress I've made.

Thanks! :smile: 

https://github.com/notpike/goodwatch/tree/jukebox

closed 2019-10-03 by notpike