travisgoodspeed/goodwatch #125: Add photo showing wire antenna installation to wiki

docprofsky, 2019-10-04:
The wiki does not have any photos showing the modification of the CA506 housing to accommodate the wire-wrap wire to connect to the wrist band for use as an antenna.
Close this issue when this is fixed.

travisgoodspeed, 2019-10-05:
Yup, we need a good assembly guide and we desperately need a good description of the antenna offerings.  Close when a wiki page exists to describe using the construction process, including the antenna.

@ea since you took those fancy photos from your last batch, may I assign this issue your way?

ea, 2019-10-05:
Sure, I was just about to build another one , currently have only one with with metal band, though. I'll write up the assembly process and you can tell me what I did wrong :)

ea, 2019-10-06:
I don't know what a proper way of tracking wiki pages would be, but ...
How's this for a construction writeup? 
https://github.com/ea/goodwatch/wiki/Construction-Guide
Photos are in my "construction photos" branch. 

I  still need to add a bit about different filter choices and antenna options.  
What are all the antenna options? I know about:
- wire antenna out the case
- SMD antenna 
- ummc connector 

Cheers

travisgoodspeed, 2019-10-07:
For the construction guide, let's begin by documenting the wire antenna woven through the rubber watchband and the wire antenna fed into the stainless steel watchband.  The chip antenna and connector can wait for now.

travisgoodspeed, 2019-10-07:
The construction write-up is great!  Please merge it into the master branch of the wiki.

One minor note: I've found that the antenna wire lasts much longer if a small slit is cut into the backplate with a Dremel tool so that the wire isn't crushed when the case is closed.

[image]

ea, 2019-10-08:
Added a bit about how i have my antenna on the plastic watch as well as your dremel slit example. Photos are now in my wiki's master branch, so you should be able to clone and merge it into yours. 

Cheres

travisgoodspeed, 2019-10-08:
Merged @ea's fork to master.

@docprofsky Please review the merged wiki page at https://github.com/travisgoodspeed/goodwatch/wiki/ConstructionGuide and close this ticket if it includes the documentation you were missing.  Feel free to make any necessary changes, or to assign the ticket to @ea or me if something is still missing.

closed 2019-10-26 by travisgoodspeed

docprofsky, 2019-10-29:
Apologies for the delay on reviewing the directions. It looks quite good and easy to follow. Thank you.