travisgoodspeed/goodwatch #137: RX Help

notpike, 2019-12-13:
Hi,

I need some help again.

I'm working on adding RX capabilities for the Jukebox app and I'm having some trouble getting the registers for the CC1101 correct. I'm using the goodwatch.py tool to check to see if the registers are working well with RX however I'm getting nothing. I'm testing with both the remote and The-Fonz script for the CC1111. Bellow are the settings I used to generate the register values with the Smart RF app. 

[image]

Bellow is a link to my fork and to use the Jukebox RX function for goodwatch.py use -J.
https://github.com/notpike/goodwatch/blob/jukebox/bin/goodwatch.py

Thank you.

travisgoodspeed, 2019-12-13:
Howdy from Hyderabad!

I'm away from my lab at the moment, but maybe I can help with some quick advice.  If not, I'll try the code when I'm back in Tennessee.

Generally, receiving is the same as transmitting, except that

(1) Your receiver bandwidth registers matter.  They should be wide enough to capture the entire packet including some drift.  If this is wrong you will still get some packets, but fewer than you expect.

(2) Your SFD/SYNC registers must match on the transmitted packet, and if this is wrong, you will get no meaningful messages.

(3) This protocol is OOK, so there might be something weird about the threshold between a zero and one.  I've not yet implemented an OOK receiver, so I don't know for sure.

(4) Some OOK receivers are *very* generous about symbol length because they recover it from the start of the packet.  The CC1101 core of the CC430 chip doesn't do this automatically, so you should double check that your symbol length is the same as the transmitter's.  If this is wrong, you will get no packets.

(5) Always debug reception with checksumming turned off, and turn it back on only after everything else is working.  If this is wrong, you will get no packets.

If this doesn't help, please tell me more about what your failures look like.  That might help diagnose the problem.

--Travis

notpike, 2019-12-14:
Thanks this helps! Figures crossed. :D

closed 2020-02-29 by travisgoodspeed