travisgoodspeed/goodwatch #147: GoodPocketWatch31

travisgoodspeed, 2020-12-24:
To date, we've been taking a GoodWatch that's designed to be worn on a wrist and fly-wiring it to an FTDI or JTAG board, then praying to half of Olympus that those tiny little wires won't snap.  This was good early on, because it forced the worn watch to be the primary development target, but it is damned inconvenient when a pull request comes in and it's too much trouble to dig out a devkit and test it.

As a solution to that problem, I'm drafting the GoodPocketWatch31, which features a USB C port and FTDI chip for convenient programming.  It re-uses the LCD from a Casio watch, but provides its own keypad.

- [ ] Schematic Complete
- [ ] PCB Layout Complete
- [ ] Gerbers Locked in Stone
- [ ] Manufacturing
- [ ] Successful Assembly

travisgoodspeed, 2020-12-24:
Haven't lain out copper yet, and it's missing some passives, but this is roughly what I'm aiming for.

[image]

travisgoodspeed, 2020-12-27:
[image]

This is a decent draft of a board that ought to be function, with an SMA edge connector for RF, USB C and FTDI for programming, and a full set of buttons.  The USB C circuit is isolated by a power gated level converter, so that the board ought to run for years on two AAAA batteries.

I'll be adding some test points and perhaps breaking out the four remaining GPIO pins before sending this off to be fabricated.

freefuel, 2022-08-25:
Travis if your still working on this please humor us Hobby R/c people and break out some of the GPIO to control servos. fwiw a speaker output would also be awesome.

travisgoodspeed, 2022-08-25:
I've been distracted by other things, but I'll leave a 0.1" proto area and some spare GPIO pins if I ever get back to it.

- [ ] Check when done.

freefuel, 2022-08-25:
how low will the radio tune?

freefuel, 2023-04-14:
Any prayer of you finishing this?

travisgoodspeed, 2023-04-19:
Currently missing is a way to mechanically attach the LCD display and its zebra strip.  I think that if those are omitted, I can finish this up pretty quickly.

freefuel, 2023-04-19:
What if it were a carrier board, with the original watch body attached and the back removed and bolted/screwed together? or perhaps a simpler board is used just for the LCD and watch case to retain the LCD.