travisgoodspeed/goodwatch #159: Can't flash chip by JTAG, because of "tilib: MSP430_OpenDevice: Unknown device (error = 5)"

vertexi, 2022-04-09:
Hi, I'm using an MSP-EXP430F5529LP launchpad to program the watch. but result in `tilib: MSP430_OpenDevice: Unknown device (error = 5)` error.

there is a detailed result:
```
MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.15.0.1 Copyright (C) 2013 TI, Inc.

Using new (SLAC460L+) API
MSP430_GetNumberOfUsbIfs
MSP430_GetNameOfUsbIf
Found FET: ttyACM0
MSP430_Initialize: ttyACM0
Firmware version is 31501001
MSP430_VCC: 3000 mV
MSP430_OpenDevice
tilib: MSP430_OpenDevice: Unknown device (error = 5)
tilib: device initialization failed
```

I thought it might be a `libmsp430.so` issue. Can you help me by providing the libmsp430.so file?

travisgoodspeed, 2022-04-09:
I'm using `libmsp430.so` from the [msp430-gcc](https://www.ti.com/tool/MSP430-GCC-OPENSOURCE) distribution.

vertexi, 2022-04-09:
Hi, there. I run the [msp430-gcc](https://www.ti.com/tool/MSP430-GCC-OPENSOURCE) installation. And copy the `msp430-gcc/bin/libmsp430.so` to the `/usr/local/lib/`. Then problem still exists.

msp430-gcc Version: 9.3.1.2 Release date: 22 Jul 2021

And there is result:
```
root@ubuntu:~/ti/msp430-gcc/bin# mspdebug tilib
MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.15.0.1 Copyright (C) 2013 TI, Inc.

Using new (SLAC460L+) API
MSP430_GetNumberOfUsbIfs
MSP430_GetNameOfUsbIf
Found FET: ttyACM0
MSP430_Initialize: ttyACM0
Firmware version is 31501001
MSP430_VCC: 3000 mV
MSP430_OpenDevice
tilib: MSP430_OpenDevice: Unknown device (error = 5)
tilib: device initialization failed
```

tahnok, 2022-04-09:
This might be dumb, but this "unknown device (error =5)" is the error I get when I swap SBWTDIO or SBWTCK (or had a poor electric connection to either)

vertexi, 2022-04-16:
Thank you all!! Finally, I can flash MCU and build the watch now.

The issue about the `unknown device (error =5)` error could attribute to the fake MCU. When I get a brand new MCU, everything works like a charm.

closed 2022-04-16 by vertexi