travisgoodspeed/goodwatch #162: [Hardware] Sourcing processors?

CalobHumble, 2022-10-27:
As of right now, the shortest lead time for compatible processors seems to be 6 weeks, from digikey [[link](https://www.digikey.com/en/products/detail/texas-instruments/cc430f6147irgct/3588864)].  This is fine, but I was hoping someone might have an alternative source for processors, like a known cheap commercial product they could be salvaged from?

Also, kudos on a super interesting project! Looking forward to putting together a class on these at my local makerspace!

CalobHumble, 2022-10-27:
@tahnok Ah, I see I could get a sample from them, but the minimum order seems to be 250 outside of a distributor.  I'll go that route, and try to get a sample to get started with, and just wait out re-order with digikey.

Thanks for the quick response!

closed 2022-10-27 by CalobHumble

tahnok, 2022-10-27:
I'm not sure you even need to ask for a sample:

https://www.ti.com/product/CC430F6147/part-details/CC430F6147IRGCR (I think
that's the right part) has a minimum of 1

CalobHumble, 2022-10-27:
I'll have to try later, I didn't make it to that screen, just saw the batch sizes on the comparisons.

CalobHumble, 2022-10-27:
Just an update for anyone looking at this thread, the parts listed above ([https://www.ti.com/product/CC430F6147/part-details/CC430F6147IRGCR](https://www.ti.com/product/CC430F6147/part-details/CC430F6147IRGCR)) is also out of stock on the official TI website.