travisgoodspeed/goodwatch #168: Basic Fab warnings and practical fixes

harrypm, 2025-07-12:
Just lost about 250USD on making 5 `goodwatch 30` boards so yeah.... 

I fixed this mess [goodwatch30_fixes.zip](https://github.com/user-attachments/files/21192181/goodwatch30_fixes.zip) would make a proper PR but the wiki really needs a 1-2-3 guide fix.

- Warnings for 0.4mm board size (fabs use 1.6mm as standard and this won't be flagged) 
- Corrections on battery terminal connections for better fab support
- Corrections on U2 pads they are too wide spaced for normal SMT fab solder pasting to cover vertical pads sometimes.
- RF antenna wrap wire THT hole to vertically thread though plastics.

[image]

[image]

[image]

https://github.com/user-attachments/assets/2129233a-3d0f-4bfa-8de7-34b96473998f

https://github.com/user-attachments/assets/5e0f2637-463a-45bf-919d-432b7a8ced3b

yuvadm, 2025-10-04:
@harrypm where did you fab your updated version? can you confirm it's a good build?

harrypm, 2025-10-04:
> @harrypm where did you fab your updated version? can you confirm it's a good build?

@yuvadm honestly I've been busy working on much more expensive projects like the [MISRC](https://github.com/Stefan-Olt/MISRC), this entire adventure was a favour for a friend I'll get around to finishing stuff properly when I throw more money at the situation, right now I need to focus on outher things financially speaking.

DominicHolmes, 2026-08-27:
@yuvadm did you ever get around to fabbing some boards? also looking into this rn

yuvadm, 2026-08-27:
@DominicHolmes yeah I did, was a bit hard fulfilling the BOM but eventually managed to get all the parts and ship a small batch of boards.